KastnerRG/riffa #3: RD_DATA widht in tx_port_channel_gate_128

dsidler, 2016-04-04:
I am reading your code base to understand how it works. So i haven't run it yet to check if it works.
i am looking at version 2.2.1 (the latest one)

In the tx_port_128.v you specify a wire:
`wire	[C_DATA_WIDTH:0] wGateData; // this is [128:0]`
this wire connects tx_port_channel_gate_128 and tx_port_monitor_128.

However, the port definition of tx_port_channel_gate:
`output [C_FIFO_DATA_WIDTH-1:0] RD_DATA,	// FIFO read data`
is not matching the `wGateData` wire, also the internal signals `rFifoData, _rFifoData` are not matching it, even though rFifoData[128] is assigned a value.

Correctly, in tx_port_monitor_128, the port definition is:
`input [C_DATA_WIDTH:0] EVT_DATA,			// Event data from tx_port_channel_gate`

I don't know it this shows up as a bug when the circuit is used, but it is clearly a bug when reading the code.

Thanks,
David

PS: if you prefer, i can create a pull request with the fix for this.

drichmond, 2016-04-04:
I think this is actually okay. The reason is that C_DATA_WIDTH = 128, and C_FIFO_DATA_WIDTH = 129. 

Does that answer your question, or am I not seeing the same issue?

dsidler, 2016-04-05:
Ok, you're right. then it was more a readability issue from my point.

Thanks, for your quick reply.

closed 2016-04-05 by dsidler