KastnerRG/riffa #7: Problems with the pci_regs.h and slab.h on L4t 23.2 (Nvidia Jetson TX1)

MrLent, 2016-05-16:
Hi,

When I compile the RIFFA driver on the target (JTX1) I have this make command error :

> make: Warning: File `circ_queue.c' has modification time 9.9e+07 s in the future
> Compiling driver for kernel: 3.10.67-g458d45c with the following values
>      NAME: 'riffa'
>    MAJNUM: '100'
> VENDOR_ID0: '10EE'
> VENDOR_ID1: '1172'
> 
> sed -i 's/#define MAJOR_NUM [^\n]*/#define MAJOR_NUM 100/g' riffa_driver.h
> sed -i 's/#define DEVICE_NAME [^\n]*/#define DEVICE_NAME "riffa"/g' riffa_driver.h
> sed -i 's/#define VENDOR_ID0 [^\n]*/#define VENDOR_ID0 0x10EE/g' riffa_driver.h
> sed -i 's/#define VENDOR_ID1 [^\n]*/#define VENDOR_ID1 0x1172/g' riffa_driver.h
> sed -i 's/#define DEBUG [^\n]*/#define DBUG 1/g' riffa_driver.h
> sed -i 's/#define DBUG [^\n]*/#define DEBUG 1/g' riffa_driver.h
> make -C /lib/modules/`uname -r`/build SUBDIRS=`pwd` modules
> make[1]: Entering directory `/usr/src/linux-headers-3.10.67-g458d45c'
> make[1]: Warning: File `/usr/src/linux-headers-3.10.67-g458d45c/arch/arm64/Makefile' has modification time 1.1e+08 s in the future
> make[2]: Warning: File `scripts/Makefile.lib' has modification time 1.1e+08 s in the future
>   CC [M]  /home/ubuntu/RIFFA_tmp/riffa_2.2.1/source/driver/linux/riffa_driver.o
> /home/ubuntu/RIFFA_tmp/riffa_2.2.1/source/driver/linux/riffa_driver.c: In function ‘fpga_probe’:
> /home/ubuntu/RIFFA_tmp/riffa_2.2.1/source/driver/linux/riffa_driver.c:1200:76: error: ‘PCI_EXP_DEVCTL2_IDO_REQ_EN’ undeclared (first use in this function)
>   error = pcie_capability_write_dword(dev,PCI_EXP_DEVCTL2,(devctl2_result | PCI_EXP_DEVCTL2_IDO_REQ_EN | PCI_EXP_DEVCTL2_IDO_CMP_EN));
>                                                                             ^
> /home/ubuntu/RIFFA_tmp/riffa_2.2.1/source/driver/linux/riffa_driver.c:1200:76: note: each undeclared identifier is reported only once for each function it appears in
> /home/ubuntu/RIFFA_tmp/riffa_2.2.1/source/driver/linux/riffa_driver.c:1200:105: error: ‘PCI_EXP_DEVCTL2_IDO_CMP_EN’ undeclared (first use in this function)
>   error = pcie_capability_write_dword(dev,PCI_EXP_DEVCTL2,(devctl2_result | PCI_EXP_DEVCTL2_IDO_REQ_EN | PCI_EXP_DEVCTL2_IDO_CMP_EN));
>                                                                                                          ^
> /home/ubuntu/RIFFA_tmp/riffa_2.2.1/source/driver/linux/riffa_driver.c: In function ‘fpga_remove’:
> /home/ubuntu/RIFFA_tmp/riffa_2.2.1/source/driver/linux/riffa_driver.c:1445:61: error: ‘PCI_EXP_DEVCTL2_IDO_REQ_EN’ undeclared (first use in this function)
>   pcie_capability_write_dword(dev,PCI_EXP_DEVCTL2,result & ~(PCI_EXP_DEVCTL2_IDO_REQ_EN | PCI_EXP_DEVCTL2_IDO_CMP_EN));
>                                                              ^
> /home/ubuntu/RIFFA_tmp/riffa_2.2.1/source/driver/linux/riffa_driver.c:1445:90: error: ‘PCI_EXP_DEVCTL2_IDO_CMP_EN’ undeclared (first use in this function)
>   pcie_capability_write_dword(dev,PCI_EXP_DEVCTL2,result & ~(PCI_EXP_DEVCTL2_IDO_REQ_EN | PCI_EXP_DEVCTL2_IDO_CMP_EN));
>                                                                                           ^
> make[2]: *** [/home/ubuntu/RIFFA_tmp/riffa_2.2.1/source/driver/linux/riffa_driver.o] Error 1
> make[1]: *** [_module_/home/ubuntu/RIFFA_tmp/riffa_2.2.1/source/driver/linux] Error 2
> make[1]: Leaving directory `/usr/src/linux-headers-3.10.67-g458d45c'
> make: *** [riffa.ko] Error 2

One solution to solve the `#include <slab.h>` problem should be to put the include in  `circ_queue.h`.

About the `pci_regs.h` `#define`, during the driver compilation the pci `#define` in `#if LINUX_VERSION_CODE <= (KERNEL_VERSION(3,6,11))` statement are not defined although the L4T 23.2 kernel has the 3.10.67 version. 

S. P

P.S: I'm using the 2.2.1 zip version available on the website  and my GCC version is the 4.8 for arm64 architecture (compilation on the target).

drichmond, 2016-06-29:
Would you be up for trying the fix I just committed in the RIFFA 2.2.2 branch

I think it it will cleanly fix the two issues you're seeing above for all L4T distributions.

MrLent, 2016-07-01:
Yes that works although I just tried with a quick program and the last L4T version (24.1 not 23.2).
As I had the same errors with the previous driver version on L4T 24.1, I think they didn't change anything about pci_reg.h.

drichmond, 2016-07-01:
Yep. I checked the 24.1 source before I made the modifications just hoping that this problem would fix itself. 

Just to confirm, the driver now compiles out of the box without errors?

MrLent, 2016-07-01:
Yes, any modification to do.

drichmond, 2016-07-18:
Closing issue, will fix in RIFFA 2.2.2

closed 2016-07-18 by drichmond