KastnerRG/riffa #8: bus master not enabled problem

intru91, 2016-05-20:
Hi,

I'm having issues with riffa while using it. 
Sometimes when the computer boots up, riffa is not detected by linux (i use an altera DE4).
When i run `python -c "import riffa;print riffa.fpga_list()"` I get `num fpgas: 0`

To debug it i found in the dmesg log those lines : 
```
797:[    5.507040] riffa: found FPGA with name: 0000:02:00.00
798:[    5.507046] riffa: vendor id: 0x1172
799:[    5.507049] riffa: device id: 0x0004
800:[    5.507054] riffa: BAR 0 address: df100000
801:[    5.507057] riffa: BAR 0 length: 1024 bytes
802:[    5.507149] riffa: MSI setup on irq 135
803:[    5.507154] riffa: PCIE_EXP_DEVCTL register: 2930
804:[    5.507159] riffa: PCIE_EXP_DEVCTL2 register: 0
805:[    5.507165] riffa: PCIE_EXP_LNKCTL register: 10810040
806:[    5.507169] riffa: number of channels: 0
807:[    5.507171] riffa: bus interface width: 0
808:[    5.507173] riffa: bus master enabled: 0
809:[    5.507175] riffa: negotiated link width: 0
810:[    5.507177] riffa: negotiated link rate: 0 MTs
811:[    5.507179] riffa: max downstream payload: 128 bytes
812:[    5.507181] riffa: max upstream payload: 128 bytes
813:[    5.507183] riffa: bus master not enabled!
```

So it seems like their is an issue with the PCIE Controller.
The weirdest thing is that this problem only appear on some synthesis. Timequest doesn't bring any timing issue in the synthesized design.

Did you ever encounter this error ?

drichmond, 2016-05-20:
That’s a very weird issue, one I have not seen before. 

Does the error depend on the bitstream, or can a single bitstream work/fail after subsequent resets?

Would you be willing to run signal tap for me? 

Dustin

intru91, 2016-05-20:
One bitstream seems to keep failing over and over.

I tried debugging using signal tap but it changes the design which sometimes eliminates the problem...

What part of the design are you interested in ?

drichmond, 2016-05-20:
I'm interested in the RXR and TXC interface of the engine layer. 

You're using RIFFA 2.2.1, right?

intru91, 2016-05-20:
I'll try to see if i can reproduce it with signal tap on those interfaces.

I'm using RIFFA 2.2.0

Also, i have noticed that quartus is detecting some nodes to be clock which are apparently not ones 👍 
```
PCI:pci_inst|riffa_wrapper_de4:riffa|riffa:riffa_inst|channel:channels[0].channel|channel_64:channel|rx_port_64:rxPort|async_fifo_fwft:mainFifo|async_fifo:fifo|rd_ptr_empty:rdPtrEmpty|rRdPtr[8]

PCI:pci_inst|riffa_wrapper_de4:riffa|riffa:riffa_inst|channel:channels[0].channel|channel_64:channel|tx_port_64:txPort|tx_port_channel_gate_64:gate|async_fifo:fifo|rd_ptr_empty:rdPtrEmpty|rRdPtr[1]
```
Could that explain part of the problem ?

drichmond, 2016-05-20:
You should try using RIFFA 2.2.1. I fixed multiple bugs between 2.2.0 and 2.2.1, at least one of which appeared only on the DE4. 

And FWIW, those wires are just weird. Xilinx and Altera both complain about them.

intru91, 2016-05-20:
I'll be trying to upgrade to 2.2.1 to see if it fixes the problem. 

If i see no difference, i will start monitoring the RXR and TXC interfaces. What should I be looking for in these ?

intru91, 2016-05-23:
After upgrading to 2.2.1, my first synthesis is working without problems. 

I'll keep you informed if the issue reappear in later synthesis.

drichmond, 2016-05-23:
Cool. If you can update me (in like a week), that would be awesome.

intru91, 2016-05-26:
Ok after some synthesis, the problem reappeared.

I've been able to insert signal tap on RXR and TXC. From what i understand, RXR_DATA_VALID and TXC_DATA_VALID are never assert during the boot process. What should I be looking for ?

drichmond, 2016-05-26:
When the host boots or restarts, the first read on the RXR (RX Request) interface should read the information register in registers.v. The response is written to the TXC interface (TX Completion).

RXR_DATA_VALID and TXC_DATA_VALID should both assert during the boot process. Can you confirm whether or not this is happening?

The next step is to understand what data is being transmitted and why it is not formatted correctly. 

A slightly different question — Are you using a Xilinx or an Altera device?

Dustin

drichmond, 2016-06-17:
@intru91, is this issue still happening?

intru91, 2016-09-05:
It is still happening, but way less often than before (1 time out of 20).

zulaffendy, 2019-04-17:
soo is there any solution to this?