KastnerRG/riffa #11: CHNL_TX_DATA_REN up for extra cycle during 16 beat transfer

mjacobsen, 2016-07-12:
version 2.2.1
gen 3 altera de5
linux

[image]

[Archive.zip](https://github.com/drichmond/riffa/files/359633/Archive.zip)

drichmond, 2016-07-18:
Yes, this is a "bug". 

The root cause of this seems to be that the CHNL_TX signal is captured in a register (rChnlTx) inside of tx_port_channel_gate_*.v. This is an issue, because it causes an additional cycle of latency in the tx_port_channel_gate state machine. Because the state machine controls CHNL_TX_REN, the fastest that CHNL_TX_REN can drop is two cycles after CHNL_TX is dropped. 

(lines 100, 107, 178, in tx_port_channel_gate_128.v are most relevant)

We have a couple of options: 

1. Remove the latency by removing the register in the path. Thus, we'd have a 1 cycle latency (REN would drop after CHNL_TX). Haven't thought of any issues with this yet (aside from timing)
2. Check rChnlTx on CHNL_TX_DATA_REN (line 97). Again, no visible issues here aside from timing (I like this solution less than 1). 
3. Raise ACK when the state machine completes the transaction cleanup. In this case, ACK acknowledges any changes in CHNL_TX. This doesn't add any latency to the user, requires no signal modifications. On the downside (and these are large), we'd probably want to mirror the change in CHNL_RX, and it would change what is a fairly stable interface. 

Feel free to chime in with opinions. I like prefer option 1 in the short term, but might switch to option 3 in RIFFA 3 (if we ever get there)

drichmond, 2016-08-09:
Fixed in upcoming RIFFA 2.2.2.

closed 2016-08-09 by drichmond