KastnerRG/riffa #15: Channel never acknowledge transaction

intru91, 2016-09-05:
Hi there,

I've noticed that after some transactions (which is a random number of times), the core stalls and stops acknowledging new transaction in the hardware side (don't know about the software side). 

The only way i found to unlock the situation is to reset the whole core which is quite inconvenient.
Even if i detect the defect by using a timeout a restarting the transfert (deasserting CHNL_TX and reasserting it), it won't acknowledge it.

Have you ever encountered this before ? And is there a fix for this ?

I'm using Riffa 2.2.1

drichmond, 2016-09-05:
Try RIFFA 2.2.2

intru91, 2016-09-05:
Should i try 2.3.0 if this one doesn't work ?

drichmond, 2016-09-05:
No

intru91, 2016-09-06:
So 2.2.2 does the same thing.
I've tried to track down the problem, where i'm at is SG_DATA_EMPTY is never deasserted in sgTXFifo. I'm following this lead to see where the core is stalled.

Do you have any pointer to where I should look at ?

intru91, 2016-09-06:
I'm switch to a different motherboard, and the issue isn't showing up...

drichmond, 2016-09-06:
Odd. Have you tried updating the bios on the failing motherboard?

intru91, 2016-09-07:
Yes, the bios is up to date.
But we had a lot of problems with this motherboard's PCIE interface.

drichmond, 2016-09-07:
What's the make/model?

drichmond, 2016-09-07:
Also, if you're interested in digging deeper and you think the MoBo might
be the problem you might want to scope the RXC_META_EP and RXR_META_EP
signals in the engine layer.

Dustin