KastnerRG/riffa #20: Orientation questions about the project

Piedone, 2017-07-27:
This project looks like something created just for our [Hastlayer project](https://hastlayer.com)!

Hastlayer automatically converts .NET programs into equivalent hardware descriptions and seamlessly swaps out software method calls with hardware executions (it's a bit like a programmer-focused HLS tool). We need a (preferably) multi-platform (as in OS and FPGA) way of supporting high-performance PCIe FPGAs and have an FPGA-side and host PC-side framework for communication where we can plug the IP cores generated by Hastlayer into and the rest is taken care of for us.

Let me have some questions:

1. Do I understand correctly that RIFFA seems suitable for our use-case?
2. Is the project kept up-to-date and alive? I see the last release was in last August. It also mentioned Vivado 2014.4.
3. If I understand correctly RIFFA utilizes the system RAM to exchange data between software and the FPGA. Does RIFFA help in any way for the FPGA logic to use memory directly for computations, not just for data exchange? (Talking about DDR RAM, not BRAM/distributed RAM; this can be either the system memory or on-board RAM.) I.e. if for example the FPGA logic wants to store 1GB of data in some memory, will RIFFA help do that, or it's still up to the developer to implement the necessary memory interface for the specific board's on-board RAM?
4. Why shouldn't we use RIFFA? :)

drichmond, 2017-07-28:
Yes - we could work for your application. I don't know of anyone using RIFFA with Partial Reconfiguration so you may run into issues there (but hopefully not)

2. It's alive. However, my time is limited (see - I haven't been able to reply recently). I'm honestly looking for someone to take over day-to-day development
3. Just data exchange, but we do have SOME ability to load data into a memory (see CHNL_RX_OFFSET)
4. See #2 - My time is limited.

Piedone, 2017-07-28:
Thank you for your answers. Actually we don't utilize partial reconfiguration, so that's no problem.

3. Basic question, but where can I find references to this (there doesn't seem to be any `CHNL_RX_OFFSET` neither in the source files nor in the docs)? Did you mean `CHNL_RX_OFF` or `SIG_CHNL_LENGTH_W`?

drichmond, 2017-07-28:
>    1. Basic question, but where can I find references to this (there

My bad - I meant

CHNL_RX_OFF

I'd be happy to talk more about the details of your project and offer
guidance if you need - just ping me over email

Piedone, 2017-07-28:
Thank you Dustin!