KastnerRG/riffa #21: riffa: BAR 0 incorrect length

loiron, 2017-08-04:
Hi,

I'm having a problem with the riffa driver. I installed without any errors, but when is loaded in the system, I have this error:

[    7.405537] riffa: module verification failed: signature and/or required key missing - tainting kernel
[    7.405776] riffa: found FPGA with name: 0000:01:00.00
[    7.405780] riffa: vendor id: 0x10EE
[    7.405782] riffa: device id: 0x6018
[    7.405785] riffa: BAR 0 address: fe600000
**[    7.405788] riffa: BAR 0 length: 1048576 bytes**
**[    7.405790] riffa: BAR 0 incorrect length**

I don't know why the length is not 1024... I'm using Ubuntu 16.04 with the kernel 4.4.0-31-generic. I tried with all the versions of the riffa driver, and with other kernels, but I couldn' solve the problem.
Any clue on what is going on?

Thank you very much,
Lois

drichmond, 2017-08-04:
Can you install the debug driver?

make debug; sudo make install

loiron, 2017-08-06:
I did that, but when executing **dmesg | grep riffa** I see the exact same information than before.

Thanks,
Lois

manbhagss, 2017-08-12:
Hi loiron,

It can happen because of many reasons, however, most frequent is that the constraints do not provide any connections to the PCIe TX and RX pins. Can you check that ?