KastnerRG/riffa #26: Linux Driver Compilation Errors

promach, 2018-01-24:
During linux driver compilation, I have such following errors [https://gist.github.com/anonymous/8c2f7fd050696f6986ba22750c57fe67](https://gist.github.com/anonymous/8c2f7fd050696f6986ba22750c57fe67)

`Compiling driver for kernel: 4.10.0-42-generic with the following values
     NAME: 'riffa'
   MAJNUM: '100'
VENDOR_ID0: '10EE'
VENDOR_ID1: '1172'

sed -i 's/#define MAJOR_NUM [^\n]*/#define MAJOR_NUM 100/g' riffa_driver.h
sed -i 's/#define DEVICE_NAME [^\n]*/#define DEVICE_NAME "riffa"/g' riffa_driver.h
sed -i 's/#define VENDOR_ID0 [^\n]*/#define VENDOR_ID0 0x10EE/g' riffa_driver.h
sed -i 's/#define VENDOR_ID1 [^\n]*/#define VENDOR_ID1 0x1172/g' riffa_driver.h
sed -i 's/#define DEBUG [^\n]*/#define DBUG 1/g' riffa_driver.h
sed -i 's/#define DBUG [^\n]*/#define DBUG 1/g' riffa_driver.h
make -C /lib/modules/`uname -r`/build SUBDIRS=`pwd` modules
make[1]: Entering directory '/usr/src/linux-headers-4.10.0-42-generic'
  CC [M]  /home/phung/Documents/fpga_overlay/riffa/driver/linux/riffa_driver.o
/home/phung/Documents/fpga_overlay/riffa/driver/linux/riffa_driver.c: In function ‘fill_sg_buf’:
/home/phung/Documents/fpga_overlay/riffa/driver/linux/riffa_driver.c:447:59: error: passing argument 5 of ‘get_user_pages’ from incompatible pointer type [-Werror=incompatible-pointer-types]
   num_pages = get_user_pages(udata, num_pages_reqd, 1, 0, pages, NULL);
                                                           ^
In file included from ./include/linux/scatterlist.h:7:0,
                 from ./include/linux/dmapool.h:14,
                 from ./include/linux/pci.h:1287,
                 from /home/phung/Documents/fpga_overlay/riffa/driver/linux/riffa_driver.c:52:
./include/linux/mm.h:1293:6: note: expected ‘struct vm_area_struct **’ but argument is of type ‘struct page **’
 long get_user_pages(unsigned long start, unsigned long nr_pages,
      ^
/home/phung/Documents/fpga_overlay/riffa/driver/linux/riffa_driver.c:447:15: error: too many arguments to function ‘get_user_pages’
   num_pages = get_user_pages(udata, num_pages_reqd, 1, 0, pages, NULL);
               ^
In file included from ./include/linux/scatterlist.h:7:0,
                 from ./include/linux/dmapool.h:14,
                 from ./include/linux/pci.h:1287,
                 from /home/phung/Documents/fpga_overlay/riffa/driver/linux/riffa_driver.c:52:
./include/linux/mm.h:1293:6: note: declared here
 long get_user_pages(unsigned long start, unsigned long nr_pages,
      ^
/home/phung/Documents/fpga_overlay/riffa/driver/linux/riffa_driver.c: At top level:
/home/phung/Documents/fpga_overlay/riffa/driver/linux/riffa_driver.c:1533:8: error: type defaults to ‘int’ in declaration of ‘DEFINE_PCI_DEVICE_TABLE’ [-Werror=implicit-int]
 static DEFINE_PCI_DEVICE_TABLE(fpga_ids) = {
        ^
/home/phung/Documents/fpga_overlay/riffa/driver/linux/riffa_driver.c:1533:1: warning: parameter names (without types) in function declaration
 static DEFINE_PCI_DEVICE_TABLE(fpga_ids) = {
 ^
/home/phung/Documents/fpga_overlay/riffa/driver/linux/riffa_driver.c:1533:1: error: function ‘DEFINE_PCI_DEVICE_TABLE’ is initialized like a variable
/home/phung/Documents/fpga_overlay/riffa/driver/linux/riffa_driver.c:1534:2: warning: braces around scalar initializer
  {PCI_DEVICE(VENDOR_ID0, PCI_ANY_ID)},
  ^
/home/phung/Documents/fpga_overlay/riffa/driver/linux/riffa_driver.c:1534:2: note: (near initialization for ‘DEFINE_PCI_DEVICE_TABLE’)
In file included from /home/phung/Documents/fpga_overlay/riffa/driver/linux/riffa_driver.c:52:0:
./include/linux/pci.h:725:2: error: field name not in record or union initializer
  .vendor = (vend), .device = (dev), \
  ^
/home/phung/Documents/fpga_overlay/riffa/driver/linux/riffa_driver.c:1534:3: note: in expansion of macro ‘PCI_DEVICE’
  {PCI_DEVICE(VENDOR_ID0, PCI_ANY_ID)},
   ^
./include/linux/pci.h:725:2: note: (near initialization for ‘DEFINE_PCI_DEVICE_TABLE’)
  .vendor = (vend), .device = (dev), \
  ^
/home/phung/Documents/fpga_overlay/riffa/driver/linux/riffa_driver.c:1534:3: note: in expansion of macro ‘PCI_DEVICE’
  {PCI_DEVICE(VENDOR_ID0, PCI_ANY_ID)},
   ^
./include/linux/pci.h:725:12: error: invalid initializer
  .vendor = (vend), .device = (dev), \
            ^
/home/phung/Documents/fpga_overlay/riffa/driver/linux/riffa_driver.c:1534:3: note: in expansion of macro ‘PCI_DEVICE’
  {PCI_DEVICE(VENDOR_ID0, PCI_ANY_ID)},
   ^
./include/linux/pci.h:725:12: note: (near initialization for ‘DEFINE_PCI_DEVICE_TABLE’)
[... 229 lines cut ...]
/home/phung/Documents/fpga_overlay/riffa/driver/linux/riffa_driver.c:1539:26: error: ‘fpga_ids’ undeclared here (not in a function)
 MODULE_DEVICE_TABLE(pci, fpga_ids);
                          ^
./include/linux/module.h:213:21: note: in definition of macro ‘MODULE_DEVICE_TABLE’
 extern const typeof(name) __mod_##type##__##name##_device_table  \
                     ^
/home/phung/Documents/fpga_overlay/riffa/driver/linux/riffa_driver.c:1533:8: warning: ‘DEFINE_PCI_DEVICE_TABLE’ defined but not used [-Wunused-function]
 static DEFINE_PCI_DEVICE_TABLE(fpga_ids) = {
        ^
In file included from /home/phung/Documents/fpga_overlay/riffa/driver/linux/riffa_driver.c:46:0:
./include/linux/module.h:213:27: error: ‘__mod_pci__fpga_ids_device_table’ aliased to undefined symbol ‘fpga_ids’
 extern const typeof(name) __mod_##type##__##name##_device_table  \
                           ^
/home/phung/Documents/fpga_overlay/riffa/driver/linux/riffa_driver.c:1539:1: note: in expansion of macro ‘MODULE_DEVICE_TABLE’
 MODULE_DEVICE_TABLE(pci, fpga_ids);
 ^
cc1: some warnings being treated as errors
scripts/Makefile.build:294: recipe for target '/home/phung/Documents/fpga_overlay/riffa/driver/linux/riffa_driver.o' failed
make[2]: *** [/home/phung/Documents/fpga_overlay/riffa/driver/linux/riffa_driver.o] Error 1
Makefile:1524: recipe for target '_module_/home/phung/Documents/fpga_overlay/riffa/driver/linux' failed
make[1]: *** [_module_/home/phung/Documents/fpga_overlay/riffa/driver/linux] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.10.0-42-generic'
Makefile:94: recipe for target 'riffa.ko' failed
make: *** [riffa.ko] Error 2
`

marzoul, 2018-01-24:
I fixed these compilation issues in my fork:
https://github.com/marzoul/riffa/
With my fixes the Riffa driver compiles and works fine with Linux up to 4.14.
There is a pull request to merge these fixes into main Riffa repo:
https://github.com/KastnerRG/riffa/pull/18
But the main Riffa author has a life and said he can't merge without proper testing etc. Please give it a try and say how it works on your side.

promach, 2018-02-08:
@marzoul 

I tried your repo, but I am facing the following problem. It seems to me that Xillybus are also using the same file as RIFFA.

> Selecting previously unselected package riffa.
(Reading database ... 396071 files and directories currently installed.)
Preparing to unpack .../riffa_20180208-1_amd64.deb ...
Unpacking riffa (20180208-1) ...
dpkg: error processing archive /home/phung/Documents/fpga_overlay/riffa_marzoul/driver/linux/riffa_20180208-1_amd64.deb (--install):
 trying to overwrite '/lib/modules/4.13.0-32-generic/modules.devname', which is also in package xillybus 20180202-1
Errors were encountered while processing:
 /home/phung/Documents/fpga_overlay/riffa_marzoul/driver/linux/riffa_20180208-1_amd64.deb

The following is my modules.devname:

> \# Device nodes to trigger on-demand module loading.
cuse cuse c10:203
autofs4 autofs c10:235
btrfs btrfs-control c10:234
userio userio c10:240
vfio vfio/vfio c10:196
hci_vhci vhci c10:137
uhid uhid c10:239
vhost_net vhost-net c10:238
vhost_vsock vhost-vsock c10:241
snd_timer snd/timer c116:33
snd_seq snd/seq c116:1

promach, 2018-02-09:
@marzoul 

I have a temporary workaround by uninstalling xillybus and reinstalling riffa linux driver.

However,  ls /dev/riffa  returns me device not found error.

So, I tried to use the pre-generated bitstream at [https://github.com/KastnerRG/riffa/tree/master/fpga/altera/de4/DE4Gen2x8If128/bit](https://github.com/KastnerRG/riffa/tree/master/fpga/altera/de4/DE4Gen2x8If128/bit) and it still results in the device file not found.

It seems to me that probably my riffa linux driver installation is not okay.

Could you advise ?

promach, 2018-02-27:
@marzoul I fixed the previous issue above and can run the demo c code without any problem.

However, when I compile the c code again today, I have the following error:

`phung@UbuntuHW15:~/Documents/fpga_overlay/riffa/c_c++/linux/x64/test_apps$ gcc testutil.c -o testutil
testutil.c: In function ‘main’:
testutil.c:190:9: warning: ‘return’ with no value, in function returning non-void
         return;
         ^
testutil.c:299:7: warning: ‘return’ with no value, in function returning non-void
       return;
       ^
testutil.c:388:7: warning: ‘return’ with no value, in function returning non-void
       return;
       ^
testutil.c:477:6: warning: ‘return’ with no value, in function returning non-void
      return;
      ^
testutil.c:566:6: warning: ‘return’ with no value, in function returning non-void
      return;
      ^
/tmp/ccrfq3ID.o: In function `main':
testutil.c:(.text+0xbf): undefined reference to `fpga_list'
testutil.c:(.text+0x260): undefined reference to `fpga_open'
testutil.c:(.text+0x2a1): undefined reference to `fpga_reset'
testutil.c:(.text+0x2b0): undefined reference to `fpga_close'
testutil.c:(.text+0x386): undefined reference to `fpga_open'
testutil.c:(.text+0x3f7): undefined reference to `fpga_close'
testutil.c:(.text+0x465): undefined reference to `fpga_close'
testutil.c:(.text+0x53b): undefined reference to `fpga_send'
testutil.c:(.text+0x5ab): undefined reference to `fpga_recv'
testutil.c:(.text+0x877): undefined reference to `fpga_close'
testutil.c:(.text+0x94d): undefined reference to `fpga_open'
testutil.c:(.text+0xa0d): undefined reference to `fpga_close'
testutil.c:(.text+0xab1): undefined reference to `fpga_close'
testutil.c:(.text+0xbb1): undefined reference to `fpga_send'
testutil.c:(.text+0xc3e): undefined reference to `fpga_recv'
testutil.c:(.text+0xf46): undefined reference to `fpga_close'
testutil.c:(.text+0x101c): undefined reference to `fpga_open'
testutil.c:(.text+0x10d9): undefined reference to `fpga_close'
testutil.c:(.text+0x114e): undefined reference to `fpga_close'
testutil.c:(.text+0x1238): undefined reference to `fpga_send'
testutil.c:(.text+0x12c5): undefined reference to `fpga_recv'
testutil.c:(.text+0x15e5): undefined reference to `fpga_close'
testutil.c:(.text+0x171b): undefined reference to `fpga_open'
testutil.c:(.text+0x17d8): undefined reference to `fpga_close'
testutil.c:(.text+0x184d): undefined reference to `fpga_close'
testutil.c:(.text+0x192f): undefined reference to `fpga_send'
testutil.c:(.text+0x199c): undefined reference to `fpga_recv'
testutil.c:(.text+0x1c6e): undefined reference to `fpga_close'
testutil.c:(.text+0x1da4): undefined reference to `fpga_open'
testutil.c:(.text+0x1e61): undefined reference to `fpga_close'
testutil.c:(.text+0x1ed6): undefined reference to `fpga_close'
testutil.c:(.text+0x1faa): undefined reference to `fpga_send'
testutil.c:(.text+0x202f): undefined reference to `fpga_recv'
testutil.c:(.text+0x233c): undefined reference to `fpga_close'
collect2: error: ld returned 1 exit status
phung@UbuntuHW15:~/Documents/fpga_overlay/riffa/c_c++/linux/x64/test_apps$`

When I run "sudo make load" as in [https://github.com/farhanrahman/riffa/wiki/Setup-Guide#installing-the-drivers](https://github.com/farhanrahman/riffa/wiki/Setup-Guide#installing-the-drivers) , my computer hanged and blacked out suddenly to this "BIOS error page, [https://i.imgur.com/BOslOUn.jpg](https://i.imgur.com/BOslOUn.jpg) and it prints out the following:

> 928-Fatal PCIe error.
>     PCIe error detected Slot 4
>     Completion timeout
> 
> 929-Fatal MCA error.
>     CB0/LLC0 error detected CPU 0
>     Memory hierarchy error

However, after I reboot my computer successfully and run "lspci" , I have the following which makes me wonder why loading the driver can cause such fatal errors. besides, I still can't compile my c code until now.

> phung@UbuntuHW15:~/Documents/fpga_overlay/riffa/c_c++/linux/x64/test_apps$ lspci | grep Altera
> 03:00.0 Unassigned class [ff00]: Altera Corporation Device 0004 (rev 01)

promach, 2018-02-28:
The hardware detection issue seems to be intermittent.

Now, my computer can detect my Altera DE4.

However, using the original demo loopback c and verilog code, I have the following:

> phung@UbuntuHW15:\~/Documents/fpga_overlay/riffa/c_c++/linux/x64/test_apps$ make all
> cc  -g   -c -o testutil.o testutil.c
> testutil.c: In function ‘main’:
> testutil.c:168:8: warning: ‘return’ with no value, in function returning non-void
>         return;
>         ^
> testutil.c:268:7: warning: ‘return’ with no value, in function returning non-void
>        return;
>        ^
> testutil.c:357:7: warning: ‘return’ with no value, in function returning non-void
>        return;
>        ^
> testutil.c:446:6: warning: ‘return’ with no value, in function returning non-void
>       return;
>       ^
> testutil.c:535:6: warning: ‘return’ with no value, in function returning non-void
>       return;
>       ^
> cc -g  testutil.o  -o testutil  -lriffa
> phung@UbuntuHW15:\~/Documents/fpga_overlay/riffa/c_c++/linux/x64/test_apps$ ./testutil 2 0 1 10000
> Running bandwidth test from 4 up to 10000 words
> Test 0: words sent: 0
> recvBuffer[10001]: 54065, expected 0
> phung@UbuntuHW15:\~/Documents/fpga_overlay/riffa/c_c++/linux/x64/test_apps$ ./testutil 2 0 1 1000
> Running bandwidth test from 4 up to 1000 words
> Test 0: words sent: 0
> recvBuffer[1001]: 126065, expected 0
> phung@UbuntuHW15:\~/Documents/fpga_overlay/riffa/c_c++/linux/x64/test_apps$ ./testutil 2 0 1 999
> Running bandwidth test from 4 up to 999 words
> Test 0: words sent: 0
> recvBuffer[1001]: 126065, expected 0
> phung@UbuntuHW15:\~/Documents/fpga_overlay/riffa/c_c++/linux/x64/test_apps$ ./testutil 0
> Number of devices: 1
> 0: id:0
> 0: num_chnls:1
> 0: name:0000:06:00.00
> 0: vendor id:1172
> 0: device id:0004
> phung@UbuntuHW15:\~/Documents/fpga_overlay/riffa/c_c++/linux/x64/test_apps$ ./testutil 1
> Usage: ./testutil 1 <fpga id>
> phung@UbuntuHW15:\~/Documents/fpga_overlay/riffa/c_c++/linux/x64/test_apps$ ./testutil 1 0
> phung@UbuntuHW15:\~/Documents/fpga_overlay/riffa/c_c++/linux/x64/test_apps$ ./testutil 2 0 1 999
> Running bandwidth test from 4 up to 999 words
> Test 0: words sent: 0
> recvBuffer[1001]: 126065, expected 0
> phung@UbuntuHW15:\~/Documents/fpga_overlay/riffa/c_c++/linux/x64/test_apps$ 

I have also verified using golden bitstream from [https://github.com/KastnerRG/riffa/blob/master/fpga/altera/de4/DE4Gen2x8If128/bit/DE4Gen2x8If128.sof](https://github.com/KastnerRG/riffa/blob/master/fpga/altera/de4/DE4Gen2x8If128/bit/DE4Gen2x8If128.sof) and it resulted in the same data integrity check error.

promach, 2018-03-16:
@marzoul I am using your patch with linux kernel 4.13

However, I am facing intermittent hardware detection issue which seems to affect the driver autoload during reboot/startup.  Could you advise ?

> phung@UbuntuHW15:~$ lsmod | grep riffa
> phung@UbuntuHW15:~$ sudo lspci -vvv | grep -A 43 Altera
> 03:00.0 Unassigned class [ff00]: Altera Corporation Device 0004 (rev 01)
> 	Subsystem: Altera Corporation Device 0004
> 	Physical Slot: 4
> 	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR+ FastB2B- DisINTx-
> 	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
> 	Latency: 0, Cache Line Size: 64 bytes
> 	Interrupt: pin A routed to IRQ 11
> 	Region 0: Memory at ef300000 (32-bit, non-prefetchable) [size=1K]
> 	Capabilities: [50] MSI: Enable- Count=1/1 Maskable- 64bit+
> 		Address: 0000000000000000  Data: 0000
> 	Capabilities: [78] Power Management version 3
> 		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
> 		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
> 	Capabilities: [80] Express (v2) Endpoint, MSI 00
> 		DevCap:	MaxPayload 256 bytes, PhantFunc 0, Latency L0s <64ns, L1 <1us
> 			ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ FLReset-
> 		DevCtl:	Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
> 			RlxdOrd- ExtTag+ PhantFunc- AuxPwr- NoSnoop+
> 			MaxPayload 256 bytes, MaxReadReq 1024 bytes
> 		DevSta:	CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr- TransPend-
> 		LnkCap:	Port #1, Speed 5GT/s, Width x8, ASPM L0s, Exit Latency L0s unlimited, L1 unlimited
> 			ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp-
> 		LnkCtl:	ASPM Disabled; RCB 64 bytes Disabled- CommClk+
> 			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
> 		LnkSta:	Speed 5GT/s, Width x8, TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
> 		DevCap2: Completion Timeout: Range ABCD, TimeoutDis+, LTR-, OBFF Not Supported
> 		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR-, OBFF Disabled
> 		LnkCtl2: Target Link Speed: 5GT/s, EnterCompliance- SpeedDis-
> 			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
> 			 Compliance De-emphasis: -6dB
> 		LnkSta2: Current De-emphasis Level: -3.5dB, EqualizationComplete-, EqualizationPhase1-
> 			 EqualizationPhase2-, EqualizationPhase3-, LinkEqualizationRequest-
> 	Capabilities: [100 v1] Virtual Channel
> 		Caps:	LPEVC=0 RefClk=100ns PATEntryBits=1
> 		Arb:	Fixed- WRR32- WRR64- WRR128-
> 		Ctrl:	ArbSelect=Fixed
> 		Status:	InProgress-
> 		VC0:	Caps:	PATOffset=00 MaxTimeSlots=1 RejSnoopTrans-
> 			Arb:	Fixed- WRR32- WRR64- WRR128- TWRR128- WRR256-
> 			Ctrl:	Enable+ ID=0 ArbSelect=Fixed TC/VC=01
> 			Status:	NegoPending- InProgress-
> 
> 05:00.0 VGA compatible controller: NVIDIA Corporation GK107GL [Quadro K2000] (rev a1) (prog-if 00 [VGA controller])
> 	Subsystem: Hewlett-Packard Company GK107GL [Quadro K2000]
> 	Physical Slot: 2
> phung@UbuntuHW15:~$

marzoul, 2018-03-16:
@promach I don't know what could happen. The riffa driver has always reliably detected my pre-programmed board when booting. But I tried to reboot the machine the least possible because my machine took ages to boot.
Maybe there is a link to pull request https://github.com/KastnerRG/riffa/pull/17 ? But they don't indicate how bad the crash is.
On my side I've faced very rare crash issues where the system logs quite clearly indicated the riffa driver was the cause. This was when using a Xilinx VC709 board, if vendor matters. Sufficiently rare (2-3 months between crashes) to prevent investigating, though. I didn't try https://github.com/KastnerRG/riffa/pull/17.

marzoul, 2018-03-16:
@promach Even if the driver is not loaded at boot, is the board detected reliably when you axplicitly ask the kernel for PCIe devices rescan after boot ? Could be an easy workaround.

promach, 2018-03-19:
@marzoul 

I have solved this. Driver will not be auto-loaded even there is a minor difference in linux kernel version, in my case 4.13.0.36  vs 4.13.0.37

closed 2018-03-19 by promach

qingcai52, 2021-06-24:
@marzoul @promach  an error occurred in executing sudo make command, I don't konw why，in ubuntu 20.04
[image]