KastnerRG/riffa #27: root scope declaration is not allowed in verilog 95/2K mode

promach, 2018-02-02:
> ERROR: [VRFC 10-1342] root scope declaration is not allowed in verilog 95/2K mode [../../../../../../../riffa_hdl/functions.vh:44]
ERROR: [VRFC 10-1342] root scope declaration is not allowed in verilog 95/2K mode [../../../../../../../riffa_hdl/functions.vh:60]
ERROR: [VRFC 10-1342] root scope declaration is not allowed in verilog 95/2K mode [../../../../../../../riffa_hdl/tlp.vh:213]
ERROR: [VRFC 10-1342] root scope declaration is not allowed in verilog 95/2K mode [../../../../../../../riffa_hdl/tlp.vh:230]

I am using Vivado version 2017.2 to compile the design. However, I got into the above errors.

I have tried both solution (introducing module wrapper as well as setting the file type as systemverilog) as in [https://stackoverflow.com/questions/44979043/vivado-sim-error-root-scope-declaration-is-not-allowed-in-verilog-95-2k-mode](https://stackoverflow.com/questions/44979043/vivado-sim-error-root-scope-declaration-is-not-allowed-in-verilog-95-2k-mode) , but they do not solve the above errors.

Jzone315, 2018-11-22:
@promach 
Did you fixed the issue? i am using riffa in vivado 2018.2, synthesized with the same issue, i am stuck here.Thanks a lot for your help.

promach, 2018-11-22:
@Jzone315 No, I am now using Quartus with Altera DE4

By the way, please have a look at https://github.com/KastnerRG/riffa/pull/31

YiSyuanChen, 2019-02-14:
@promach I'm facing exactly same problem. However, I'm wondering if the simulation is really workable? It seems that there is no usual testbench file for simulation. Does that means I can only verify the design by generate bitstream then load to FPGA? Any comment would be extremely appreciated.

promach, 2019-02-15:
@YiSyuanChen You could probably write your own simulation testbench, but for me, I use waveforms captured using ILA (some people call it as chipscope) which you should be very familiar with

YiSyuanChen, 2019-02-15:
@promach Thanks for your rapid response, and you give a really useful information!! I'll check it in this way.