KastnerRG/riffa #29: Bandwidth drops for linux kernel 4.15.5

promach, 2018-02-28:
When I upgrade to linux kernel 4.15.5 , testutil check is only giving me around 1.5Gbps for very large sample size.

> phung@UbuntuHW15:\~/Documents/fpga_overlay/riffa/c_c++/linux/x64/test_apps$ ./testutil 2 0 0 9999999999
> Test 0: words sent: 1048576
> Test 0: words recv: 1048576
> Test 1: words sent: 1048576
> Test 1: words recv: 1048576
> send bw: 1474.287238
> recv bw: 1507.535028
> Test 2: words sent: 1048576
> Test 2: words recv: 1048576
> send bw: 1447.335230
> recv bw: 1542.593983
> Test 3: words sent: 1048576
> Test 3: words recv: 1048576
> send bw: 1498.592916
> recv bw: 1565.649247
> Test 4: words sent: 1048576
> Test 4: words recv: 1048576
> send bw: 1502.262083
> recv bw: 1548.852252
> Test 5: words sent: 1048576
> Test 5: words recv: 1048576
> send bw: 1503.840090
> recv bw: 1564.508623
> Test 6: words sent: 1048576
> Test 6: words recv: 1048576
> send bw: 1498.462205
> recv bw: 1551.229723
> Test 7: words sent: 1048576
> Test 7: words recv: 1048576
> send bw: 1444.293332
> recv bw: 1548.852252
> Test 8: words sent: 1048576
> Test 8: words recv: 1048576
> send bw: 1443.807814
> recv bw: 1518.595349
> Test 9: words sent: 1048576
> Test 9: words recv: 1048576
> send bw: 1436.924488
> recv bw: 1534.053861
> Test 10: words sent: 1048576
> Test 10: words recv: 1048576
> send bw: 1470.627391
> recv bw: 1536.935873
> Test 11: words sent: 1048576
> Test 11: words recv: 1048576
> send bw: 1467.110947
> recv bw: 1547.735963
> Test 12: words sent: 1048576
> Test 12: words recv: 1048576
> send bw: 1475.806991
> recv bw: 1501.736817
> Test 13: words sent: 1048576
> Test 13: words recv: 1048576
> send bw: 1449.288779
> recv bw: 1538.587604
> Test 14: words sent: 1048576
> Test 14: words recv: 1048576
> send bw: 1456.291361
> recv bw: 1550.529710
> Test 15: words sent: 1048576
> Test 15: words recv: 1048576
> send bw: 1451.860829
> recv bw: 1530.227949
> Test 16: words sent: 1048576
> Test 16: words recv: 1048576
> send bw: 1459.384063
> recv bw: 1555.161508
> Test 17: words sent: 1048576
> Test 17: words recv: 1048576
> send bw: 1431.894414
> recv bw: 1516.986241
> Test 18: words sent: 1048576
> Test 18: words recv: 1048576
> send bw: 1453.334674
> recv bw: 1501.736817
> Test 19: words sent: 1048576
> Test 19: words recv: 1048576
> send bw: 1447.335230
> recv bw: 1539.138970
> Test 20: words sent: 1048576
> Test 20: words recv: 1048576
> send bw: 1441.869004
> recv bw: 1503.313719
> Test 21: words sent: 1048576
> Test 21: words recv: 1048576
> send bw: 1457.279598
> recv bw: 1548.852252
> Test 22: words sent: 1048576
> Test 22: words recv: 1048576
> send bw: 1454.318902
> recv bw: 1557.558403
> Test 23: words sent: 1048576
> Test 23: words recv: 1048576
> send bw: 1445.751846
> recv bw: 1507.667326
> Test 24: words sent: 1048576
> Test 24: words recv: 1048576
> send bw: 1459.880114
> recv bw: 1531.319118
> Test 25: words sent: 1048576
> Test 25: words recv: 1048576
> send bw: 1469.118282
> recv bw: 1516.450630
> Test 26: words sent: 1048576
> Test 26: words recv: 1048576
> send bw: 1439.452801
> recv bw: 1561.522376
> Test 27: words sent: 1048576
> Test 27: words recv: 1048576
> send bw: 1472.644367
> recv bw: 1531.319118
> Test 28: words sent: 1048576
> Test 28: words recv: 1048576
> send bw: 1465.984229
> recv bw: 1553.474020
> Test 29: words sent: 1048576
> Test 29: words recv: 1048576
> send bw: 1503.313719
> recv bw: 1566.791535
> Test 30: words sent: 1048576
> Test 30: words recv: 1048576
> send bw: 1503.313719
> recv bw: 1564.508623
> Test 31: words sent: 1048576
> Test 31: words recv: 1048576
> send bw: 1475.300059
> recv bw: 1574.401501
> Test 32: words sent: 1048576
> Test 32: words recv: 1048576
> send bw: 1481.534079
> recv bw: 1582.668741
> Test 33: words sent: 1048576
> Test 33: words recv: 1048576
> send bw: 1477.329881
> recv bw: 1551.229723
> Test 34: words sent: 1048576
> Test 34: words recv: 1048576
> send bw: 1482.556885
> recv bw: 1559.821063
> Test 35: words sent: 1048576
> Test 35: words recv: 1048576
> send bw: 1486.276424
> recv bw: 1549.411001
> Test 36: words sent: 1048576
> Test 36: words recv: 1048576
> send bw: 1484.735043
> recv bw: 1562.232353
> Test 37: words sent: 1048576
> Test 37: words recv: 1048576
> send bw: 1480.002514
> recv bw: 1562.090306
> Test 38: words sent: 1048576
> Test 38: words recv: 1048576
> send bw: 1506.609593
> recv bw: 1566.220183
> Test 39: words sent: 1048576
> Test 39: words recv: 1048576
> send bw: 1468.615933
> recv bw: 1567.363305
> Test 40: words sent: 1048576
> Test 40: words recv: 1048576
> send bw: 1486.276424
> recv bw: 1555.161508
> Test 41: words sent: 1048576
> Test 41: words recv: 1048576
> send bw: 1502.787717
> recv bw: 1570.371955
> Test 42: words sent: 1048576
> Test 42: words recv: 1048576
> send bw: 1488.336584
> recv bw: 1558.123452
> Test 43: words sent: 1048576
> Test 43: words recv: 1048576
> send bw: 1475.300059
> recv bw: 1562.800799
> Test 44: words sent: 1048576
> Test 44: words recv: 1048576
> send bw: 1462.365440
> recv bw: 1581.066555
> Test 45: words sent: 1048576
> Test 45: words recv: 1048576
> send bw: 1471.635188
> recv bw: 1548.293906
> Test 46: words sent: 1048576
> Test 46: words recv: 1048576
> send bw: 1491.049226
> recv bw: 1536.386083
> Test 47: words sent: 1048576
> Test 47: words recv: 1048576
> send bw: 1489.497935
> recv bw: 1544.813343
> Test 48: words sent: 1048576
> Test 48: words recv: 1048576
> send bw: 1487.305790
> recv bw: 1546.064541
> Test 49: words sent: 1048576
> Test 49: words recv: 1048576
> send bw: 1499.508526
> recv bw: 1555.302298
> Test 50: words sent: 1048576
> Test 50: words recv: 1048576
> send bw: 1494.811553
> recv bw: 1561.522376
> Test 51: words sent: 1048576
> Test 51: words recv: 1048576
> send bw: 1500.556309
> recv bw: 1543.702865
> Test 52: words sent: 1048576
> Test 52: words recv: 1048576
> send bw: 1495.331986
> recv bw: 1534.739073
> Test 53: words sent: 1048576
> Test 53: words recv: 1048576
> send bw: 1447.335230
> recv bw: 1520.207874
> Test 54: words sent: 1048576
> Test 54: words recv: 1048576
> send bw: 1503.840090
> recv bw: 1564.508623
> Test 55: words sent: 1048576
> Test 55: words recv: 1048576
> send bw: 1476.441147
> recv bw: 1564.936162
> Test 56: words sent: 1048576
> Test 56: words recv: 1048576
> send bw: 1488.336584
> recv bw: 1578.016826
> Test 57: words sent: 1048576
> Test 57: words recv: 1048576
> send bw: 1457.403222
> recv bw: 1538.587604
> Test 58: words sent: 1048576
> Test 58: words recv: 1048576
> send bw: 1493.771775
> recv bw: 1551.089670
> Test 59: words sent: 1048576
> Test 59: words recv: 1048576
> send bw: 1488.336584
> recv bw: 1552.912337
> Test 60: words sent: 1048576
> Test 60: words recv: 1048576
> send bw: 1474.287238
> recv bw: 1563.227405
> Test 61: words sent: 1048576
> Test 61: words recv: 1048576
> send bw: 1488.981555
> recv bw: 1560.954860
> Test 62: words sent: 1048576
> Test 62: words recv: 1048576
> send bw: 1467.612266
> recv bw: 1538.587604
> Test 63: words sent: 1048576
> Test 63: words recv: 1048576
> send bw: 1497.417344
> recv bw: 1565.649247
> Test 64: words sent: 1048576
> Test 64: words recv: 1048576
> send bw: 1488.852516
> recv bw: 1550.529710
> Test 65: words sent: 1048576
> Test 65: words recv: 1048576
> send bw: 1490.014673
> recv bw: 1548.852252
> Test 66: words sent: 1048576
> Test 66: words recv: 1048576
> send bw: 1495.852780
> recv bw: 1548.293906
> Test 67: words sent: 1048576
> Test 67: words recv: 1048576
> send bw: 1479.492696
> recv bw: 1557.417205
> Test 68: words sent: 1048576
> Test 68: words recv: 1048576
> send bw: 1486.276424
> recv bw: 1556.288539
> Test 69: words sent: 1048576
> Test 69: words recv: 1048576
> send bw: 1487.949869
> recv bw: 1555.161508
> Test 70: words sent: 1048576
> Test 70: words recv: 1048576
> send bw: 1501.211918
> recv bw: 1557.558403
> Test 71: words sent: 1048576
> Test 71: words recv: 1048576
> send bw: 1471.761260
> recv bw: 1563.227405
> Test 72: words sent: 1048576
> Test 72: words recv: 1048576
> send bw: 1483.709231
> recv bw: 1549.411001
> Test 73: words sent: 1048576
> Test 73: words recv: 1048576
> send bw: 1501.736817
> recv bw: 1542.040139
> ^C
> phung@UbuntuHW15:\~/Documents/fpga_overlay/riffa/c_c++/linux/x64/test_apps$ uname -a
> Linux UbuntuHW15 4.15.5-041505-generic #201802261304 SMP Mon Feb 26 18:05:53 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
> phung@UbuntuHW15:\~/Documents/fpga_overlay/riffa/c_c++/linux/x64/test_apps$

marzoul, 2018-02-28:
I don't understand the title of your which says bandwidth drops. With what and compared to what ? By how much ?

What I see is that you asked for 9999999999 words = 10 billion. Not only this overflows 32 bits (see the code of the app and try `man atoi`) but it also exceeds what Riffa is capable of per tranfer (2 billions, 31 bits). So getting this cropped somewhere (to 65536 it seems) is not a surprise ;-)

Just to be clear, I didn't modify the kernel in https://github.com/KastnerRG/riffa/pull/18, I just applied the macros/functions renaming that the kernel developers did along kernel versions.

If you noticed a BW drop for all kernel versions, this makes me think of recent Spectre/Meltdown patches that suddenly got applied to ALL kernel versions. As this isolates kernel pages from other process pages (if you're using Intel CPU), the only impact this can have on perfs is a slowdown. Assuming this is the cause of course. In that case, I know there were discussions to add a kernel parameter so this this costly kernel page table isolation can be disabled, but I don't know the current status of this (if this parameter does exist, it's probably only for latest kernel anyway).

Other than that, please check your system logs when an FPGA is discovered by the kernel. It prints the negociated speed and other interesting information. Your 1400/1500 MB/s seems like PCIe Gen1 2.5GT/s 8x to me, with BW significantly lower than theoretical 2GB/s. You should check with much larger buffers to mask sync latency and to get closer to what your machine can achieve (like 100MB at least).

promach, 2018-03-01:
The bandwidth drop is compared with what I got previously for Gen 2 result which is around 3.5Gbps.

I have tried re-downloading the bitstream, rebooting my computer. Only one out of many times of retrying solved the problem. I am not sure if this relates to linux driver or kernel version.

Besides, as of why we are only receiving less data than we sent, please see [https://groups.google.com/forum/#!topic/riffa_users/z1hQ7F0vFGs](https://groups.google.com/forum/#!topic/riffa_users/z1hQ7F0vFGs) or the following:

> The reason for this strange phenomenon is that RIFFA will not be able to receive data that does not follow https://github.com/KastnerRG/riffa/blob/master/c_c%2B%2B/linux/x64/test_apps/testutil.c#L138
> 
> Besides, in http://riffa.ucsd.edu/node/10 , for fpga_recv(), the author quoted "The FPGA will specify an offset value which will determine where received data will start being written. If the amount of data plus the offset exceed the size of the data array, then the additional data will be discarded."
> 
> I am not sure why RIFFA authors decide to implement this way.
> 
> Does anyone have any idea or comments about this design decision to discard words which seems not okay to users ?

drichmond, 2018-03-01:
What do you observe when you run lspci

promach, 2018-03-01:
"lspci | grep Altera" gave me 

> 03:00.0 Unassigned class [ff00]: Altera Corporation Device 0004 (rev 01)

drichmond, 2018-03-01:
How about lspci -vvv

I need to see more than that first line

promach, 2018-03-01:
Here you go:

>  phung@UbuntuHW15:~$ lspci -vvv
> 00:00.0 Host bridge: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 DMI2 (rev 04)
> 	Subsystem: Hewlett-Packard Company Xeon E7 v2/Xeon E5 v2/Core i7 DMI2
> 	Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR+ FastB2B- DisINTx-
> 	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
> 	Interrupt: pin A routed to IRQ 0
> 	Capabilities: <access denied>
> 
> 00:01.0 PCI bridge: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 PCI Express Root Port 1a (rev 04) (prog-if 00 [Normal decode])
> 	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr+ Stepping- SERR+ FastB2B- DisINTx-
> 	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
> 	Latency: 0, Cache Line Size: 64 bytes
> 	Interrupt: pin A routed to IRQ 24
> 	Bus: primary=00, secondary=03, subordinate=03, sec-latency=0
> 	I/O behind bridge: 0000f000-00000fff
> 	Memory behind bridge: ef300000-ef3fffff
> 	Prefetchable memory behind bridge: 00000000fff00000-00000000000fffff
> 	Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort+ <SERR- <PERR-
> 	BridgeCtl: Parity+ SERR+ NoISA- VGA- MAbort- >Reset- FastB2B-
> 		PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
> 	Capabilities: <access denied>
> 	Kernel driver in use: pcieport
> 	Kernel modules: shpchp
> 
> 00:02.0 PCI bridge: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 PCI Express Root Port 2a (rev 04) (prog-if 00 [Normal decode])
> 	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr+ Stepping- SERR+ FastB2B- DisINTx-
> 	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
> 	Latency: 0, Cache Line Size: 64 bytes
> 	Interrupt: pin A routed to IRQ 25
> 	Bus: primary=00, secondary=05, subordinate=05, sec-latency=0
> 	I/O behind bridge: 0000c000-0000cfff
> 	Memory behind bridge: ee000000-ef0fffff
> 	Prefetchable memory behind bridge: 00000000d0000000-00000000e1ffffff
> 	Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort+ <SERR- <PERR-
> 	BridgeCtl: Parity+ SERR+ NoISA- VGA+ MAbort- >Reset- FastB2B-
> 		PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
> 	Capabilities: <access denied>
> 	Kernel driver in use: pcieport
> 	Kernel modules: shpchp
> 
> 00:03.0 PCI bridge: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 PCI Express Root Port 3a (rev 04) (prog-if 00 [Normal decode])
> 	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr+ Stepping- SERR+ FastB2B- DisINTx-
> 	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
> 	Latency: 0, Cache Line Size: 64 bytes
> 	Interrupt: pin A routed to IRQ 26
> 	Bus: primary=00, secondary=04, subordinate=04, sec-latency=0
> 	I/O behind bridge: 0000f000-00000fff
> 	Memory behind bridge: fff00000-000fffff
> 	Prefetchable memory behind bridge: 00000000fff00000-00000000000fffff
> 	Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- <SERR- <PERR-
> 	BridgeCtl: Parity+ SERR+ NoISA- VGA- MAbort- >Reset- FastB2B-
> 		PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
> 	Capabilities: <access denied>
> 	Kernel driver in use: pcieport
> 	Kernel modules: shpchp
> 
> 00:04.0 System peripheral: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Crystal Beach DMA Channel 0 (rev 04)
> 	Subsystem: Hewlett-Packard Company Xeon E7 v2/Xeon E5 v2/Core i7 Crystal Beach DMA Channel 0
> 	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
> 	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
> 	Latency: 0, Cache Line Size: 64 bytes
> 	Interrupt: pin A routed to IRQ 41
> 	Region 0: Memory at ef43c000 (64-bit, non-prefetchable) [size=16K]
> 	Capabilities: <access denied>
> 	Kernel driver in use: ioatdma
> 	Kernel modules: ioatdma
> 
> 00:04.1 System peripheral: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Crystal Beach DMA Channel 1 (rev 04)
[... 559 lines cut ...]
> 	Subsystem: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 QPI Ring Registers
> 	Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
> 	Status: Cap- 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
> 
> 7f:13.5 Performance counters: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 QPI Ring Performance Ring Monitoring (rev 04)
> 	Subsystem: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 QPI Ring Performance Ring Monitoring
> 	Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
> 	Status: Cap- 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
> 	Kernel driver in use: ivbep_uncore
> 
> 7f:16.0 System peripheral: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 System Address Decoder (rev 04)
> 	Subsystem: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 System Address Decoder
> 	Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
> 	Status: Cap- 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
> 
> 7f:16.1 System peripheral: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Broadcast Registers (rev 04)
[... cut ...]
> 	Status: Cap- 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
> 
> 7f:16.2 System peripheral: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Broadcast Registers (rev 04)
> 	Subsystem: Intel Corporation Xeon E7 v2/Xeon E5 v2/Core i7 Broadcast Registers
> 	Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
> 	Status: Cap- 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
> phung@UbuntuHW15:~$

drichmond, 2018-03-01:
Still can't see what I'm looking for. Try `sudo lspci -vvv | grep -A 10 altera`

promach, 2018-03-01:
Here you go :

> phung@UbuntuHW15:\~$ sudo lspci -vvv | grep -A 10 altera
> [sudo] password for phung: 
> phung@UbuntuHW15:\~$ sudo lspci -vvv | grep -A 10 altera
> phung@UbuntuHW15:\~$ sudo lspci -vvv | grep -A 10 Altera
> 03:00.0 Unassigned class [ff00]: Altera Corporation Device 0004 (rev 01)
> 	Subsystem: Altera Corporation Device 0004
> 	Physical Slot: 4
> 	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR+ FastB2B- DisINTx+
> 	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
> 	Latency: 0, Cache Line Size: 64 bytes
> 	Interrupt: pin A routed to IRQ 52
> 	Region 0: Memory at ef300000 (32-bit, non-prefetchable) [size=1K]
> 	Capabilities: [50] MSI: Enable+ Count=1/1 Maskable- 64bit+
> 		Address: 00000000fee00618  Data: 0000
> 	Capabilities: [78] Power Management version 3
> 		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
> phung@UbuntuHW15:\~$

drichmond, 2018-03-01:
> Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- SERR- <PERR- INTx-

Okay - it didn’t show up in the first 10 lines

How about `lspci -vvv | grep -A 100 Altera`

I’m looking for a line like: 

                LnkSta: Speed 2.5GT/s, Width x16, TrErr- Train- SlotClk+ DLActive+ BWMgmt+ ABWMgmt-

And while you’re at it - do you have the debug driver installed? If so - then also run the command

dmesg | grep riffa

promach, 2018-03-01:
> LnkSta: Speed 5GT/s, Width x8, TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-

Please check [my updated comment](https://github.com/KastnerRG/riffa/issues/29#issuecomment-369484068)

I have just ran "./testutil 2 0 0 55536" . [This is what I got](https://gist.github.com/anonymous/72fdf097a3b3e9e23067d9e87354be1b) for "dmesg | grep riffa".  The log still shows number of received words(32768) is less than number of words sent (55536).

promach, 2018-03-03:
The relevant code to discard additional data is highlighted in https://github.com/KastnerRG/riffa/blob/master/driver/linux/riffa_driver.c#L640-L650

Does anyone have any comments on how to trigger this c driver code segment because I have tried and I have been receiving all the words I sent after modifying the test_util c code at https://gist.github.com/promach/5fb1ddfab95d4e72033a3053735f9df1#file-testutil-c-L155 ?

promach, 2018-03-06:
I am using Altera DE4 and it seems like my intermittent PCIe hardware detection issue is related to [https://i.imgur.com/hGSuCdj.png](https://i.imgur.com/hGSuCdj.png) ?

@drichmond @mjacobsen 

However, could anyone help to guide me how to do the timing closure for this particular path which involves the PCIe hardcore IP ?

jixi2018, 2019-08-09:
@drichmond , @promach 

I downloaded the bitstream file KC705_Gen2x8If128.bit to my KC705 board, and installed the driver version 2.2.2. When I ran the testutil, the measured BW was only 425MB/s, which is far less than the expectation of 3500MB/s.

./testutil 2 0 0 30000000
words sent: 30000000
words recv: 30000000
recvBuffer[0]: 29999997
recvBuffer[1]: 29999998
recvBuffer[2]: 29999999
recvBuffer[3]: 30000000
............
send bw: 424.788240 MB/s 269.406982ms
recv bw: 429.138382 MB/s 266.676025ms

Following the previous suggestions, I ran the following command to check the PCIe link status:

$ sudo lspci -s "02:00.0" -vvv
0000:02:00.0 Memory controller: Xilinx Corporation Device 7018
	Subsystem: Xilinx Corporation Device 0007
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 64 bytes
	Interrupt: pin ? routed to IRQ 79
	NUMA node: 0
	Region 0: Memory at 92100000 (32-bit, non-prefetchable) [size=1K]
	Capabilities: [40] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0+,D1+,D2+,D3hot+,D3cold-)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [48] MSI: Enable+ Count=1/1 Maskable- 64bit+
		Address: 00000000fee07000  Data: 4025
	Capabilities: [60] Express (v2) Endpoint, MSI 00
		DevCap:	MaxPayload 256 bytes, PhantFunc 0, Latency L0s <64ns, L1 unlimited
			ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ FLReset- SlotPowerLimit 10.000W
		DevCtl:	Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
			RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+
			MaxPayload 256 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr- TransPend-
		LnkCap:	Port #0, **Speed 5GT/s, Width x8**, ASPM L0s, Exit Latency L0s unlimited, L1 unlimited
			ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp-
		LnkCtl:	ASPM Disabled; RCB 128 bytes Disabled- CommClk+
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	**Speed 5GT/s, Width x1**, TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-

LnkCap shows that the number of PCI lanes is x8, however, LnkSta only detects x1 lane configuration. The KC705 board is plugged in a PCIe3X16 slot of my PC. Multiplying the measured BW by 8 should get expected performance of 425*8 = 3400MB/s. Now the question is why the linux only detects one lane of PCIe, and how to utilize all the 8 PCIe lanes? Could you provide some suggestions?

-Jinwen