KastnerRG/riffa #32: unable to resolve clog2s?

Jzone315, 2018-11-21:
Hi:
   Sorry to bother you, i am using the xilinx adm7V3 example designs in vivado 2018.2, but when i run synthesis it complaint [Synth 8-660] unable to resolve 'clog2s'[fiffa_wrapper_adm7V3.v:164], when i tried the vivado 2017.2, it reported the same issue, so why is this error happen? i needed some help, thanks a lot.

hmaarrfk, 2019-01-18:
See this maybe: https://github.com/KastnerRG/riffa/pull/36

gxflying, 2019-08-16:
actually, this issue is all about the file including just like I had met these days.
you can try to move the line `include "functions.vh" in async_fifo_fwft.v the beginning of the file.
this issue comes when:
1. when synthesis , the file async_fifo_fwft.v is at the very front of the file parsing list
2. in the async_fifo_fwft.v, the functions.vh is included inside the module, so all the function defined in the file is locally defined, NOT globaly.   and the file will NOT be parsed twice because of the defination of the maro __FUNCTIONS_VH