KastnerRG/riffa #40: ERROR: [VRFC 10-1775] range must be bounded by constant expressions

YiSyuanChen, 2019-02-14:
Hi all, I'm trying to simulate the project for VC707 (VC707_gen1x8lf64) using Vivado 2015.4. The Vivado runs on Ubuntu 14.04. However, the simulation has errors as following:

ERROR: [VRFC 10-1342] root scope declaration is not allowed in verilog 95/2K mode [../../../../../../../riffa_hdl/functions.vh:45]
ERROR: [VRFC 10-1342] root scope declaration is not allowed in verilog 95/2K mode [../../../../../../../riffa_hdl/functions.vh:61]
...
ERROR: [VRFC 10-1775] range must be bounded by constant expressions [/home/jeff/riffa_ori/fpga/riffa_hdl/scsdpram.v:59]
ERROR: [VRFC 10-1775] range must be bounded by constant expressions [/home/jeff/riffa_ori/fpga/riffa_hdl/scsdpram.v:63]

For the first error, it seems that the functions in verilog header must be wrapped in a module. There are two more files have similar problem.

For the second error, the simulator reports error for syntax like: input [clog2s(C_DEPTH)-1:0]

I have tried to replace all clog2 and clog2s with $clog2, and wrap the functions into module, but the simulation result was very wierd. Does anyone can provide possible solution for these errors or any idea about simulation for project? Any comment would be very appreciated!

sjkklein, 2020-07-02:
I am running into this same problem. Did you ever find a solution? I am going to keep poking at it...

dxloc, 2021-12-01:
Same problem with the default simulator in Vivado 2019.2. Synthesis is still OK. Have anyone found the solution?

dxloc, 2021-12-01:
seem like if you add a parameter that is the clog2s, you can solve this error
scsdpram.v:

module scsdpram
    #(
      parameter C_WIDTH = 32,
      parameter C_DEPTH = 1024,
// Add C_ADDR_WIDTH parameter
      parameter C_ADDR_WIDTH = clog2s(C_DEPTH)
      )
    (
     input                       CLK,
     input                       RD1_EN,
//     input [clog2s(C_DEPTH)-1:0] RD1_ADDR,
     input [C_ADDR_WIDTH-1:0] RD1_ADDR,
     output [C_WIDTH-1:0]        RD1_DATA,
     input                       WR1_EN,
//     input [clog2s(C_DEPTH)-1:0] WR1_ADDR,
     input [C_ADDR_WIDTH-1:0] WR1_ADDR,
     input [C_WIDTH-1:0]         WR1_DATA
     );

showmeee, 2024-01-03:
I think run the script would solve the 2nd question