KastnerRG/riffa #43: Cannot compile the driver on Ubuntu-18.04

jixi2018, 2019-08-07:
When I ran "make" under riffa/driver/linux directory, the following errors were generated:

jinwen@xbx-jixi-09:~/projects/riffa/driver/linux$ make debug
Compiling driver for kernel: 4.18.0-22-generic with the following values
     NAME: 'riffa'
   MAJNUM: '100'
VENDOR_ID0: '10EE'
VENDOR_ID1: '1172'

sed -i 's/#define MAJOR_NUM [^\n]*/#define MAJOR_NUM 100/g' riffa_driver.h
sed -i 's/#define DEVICE_NAME [^\n]*/#define DEVICE_NAME "riffa"/g' riffa_driver.h
sed -i 's/#define VENDOR_ID0 [^\n]*/#define VENDOR_ID0 0x10EE/g' riffa_driver.h
sed -i 's/#define VENDOR_ID1 [^\n]*/#define VENDOR_ID1 0x1172/g' riffa_driver.h
sed -i 's/#define DEBUG [^\n]*/#define DBUG 1/g' riffa_driver.h
sed -i 's/#define DBUG [^\n]*/#define DEBUG 1/g' riffa_driver.h
make -C /lib/modules/`uname -r`/build SUBDIRS=`pwd` modules
make[1]: Entering directory '/usr/src/linux-headers-4.18.0-22-generic'
arch/x86/Makefile:155: CONFIG_X86_X32 enabled but no binutils support
arch/x86/Makefile:244: *** You are building kernel with non-retpoline compiler, please update your compiler..  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-4.18.0-22-generic'
Makefile:94: recipe for target 'riffa.ko' failed
make: *** [riffa.ko] Error 2

Could you advise how to fix this?

jixi2018, 2019-08-07:
I updated the gcc version to 7.4.0, the previous error was gone, but the following compile error showed:(I listed the first one to minimize the message length)

/home/projects/riffa/driver/linux/riffa_driver.c: In function ‘fill_sg_buf’:
/home/projects/riffa/driver/linux/riffa_driver.c:447:59: error: passing argument 5 of ‘get_user_pages’ from incompatible pointer type [-Werror=incompatible-pointer-types]
   num_pages = get_user_pages(udata, num_pages_reqd, 1, 0, pages, NULL);
                                                           ^~~~~
In file included from ./include/linux/scatterlist.h:8:0,
                 from ./include/linux/dmapool.h:14,
                 from ./include/linux/pci.h:1334,
                 from /home/jinwen/projects/riffa/driver/linux/riffa_driver.c:52:
./include/linux/mm.h:1484:6: note: expected ‘struct vm_area_struct **’ but argument is of type ‘struct page **’
 long get_user_pages(unsigned long start, unsigned long nr_pages,
      ^~~~~~~~~~~~~~
/home/projects/riffa/driver/linux/riffa_driver.c:447:15: error: too many arguments to function ‘get_user_pages’
   num_pages = get_user_pages(udata, num_pages_reqd, 1, 0, pages, NULL);
               ^~~~~~~~~~~~~~
In file included from ./include/linux/scatterlist.h:8:0,
                 from ./include/linux/dmapool.h:14,
                 from ./include/linux/pci.h:1334,
                 from /home/projects/riffa/driver/linux/riffa_driver.c:52:
./include/linux/mm.h:1484:6: note: declared here
 long get_user_pages(unsigned long start, unsigned long nr_pages,
      ^~~~~~~~~~~~~~

marzoul, 2019-08-08:
Hi jixi2018, your linux kernel is probably too recent. I made pull request #18 some time ago to fix this, please give it a try.

jixi2018, 2019-08-08:
Thanks. Could you let me know how to get the code in the pull request #18 via command line? I am sort of new to git.....

marzoul, 2019-08-09:
Hi @jixi2018, I think the simplest way is to switch to my small fork of Riffa that I made for that pull request. The URL is the following:
`https://github.com/marzoul/riffa.git`

jixi2018, 2019-08-14:
Issue was resolved after following the changes in the pull request #18. Closing it.

closed 2019-08-14 by jixi2018