KastnerRG/riffa #44: riffa python example have different data between TX and RX data

jixi2018, 2019-08-14:
I download the RIFFA bitstream with Gen 2.0x8 configuration into my KC705 FPGA and installed the drivers on Ubuntu 18.04, and C test worked fine with the measured bandwidth of:

send bw: 3335.055111 MB/s 114.381836ms
recv bw: 3455.153763 MB/s 110.406006ms

However, the python example showed that the data received were different from the data that were transmitted. 

$ python sampleapp.py 
TX data: 
array('I', [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47, 48, 49, 50, 51, 52, 53, 54, 55, 56, 57, 58, 59, 60, 61, 62, 63, 64, 65, 66, 67, 68, 69, 70, 71, 72, 73, 74, 75, 76, 77, 78, 79, 80, 81, 82, 83, 84, 85, 86, 87, 88, 89, 90, 91, 92, 93, 94, 95, 96, 97, 98, 99, 100, 101, 102, 103, 104, 105, 106, 107, 108, 109, 110, 111, 112, 113, 114, 115, 116, 117, 118, 119, 120, 121, 122, 123, 124, 125, 126, 127, 128, 129, 130, 131, 132, 133, 134, 135, 136, 137, 138, 139, 140, 141, 142, 143, 144, 145, 146, 147, 148, 149, 150, 151, 152, 153, 154, 155, 156, 157, 158, 159, 160, 161, 162, 163, 164, 165, 166, 167, 168, 169, 170, 171, 172, 173, 174, 175, 176, 177, 178, 179, 180, 181, 182, 183, 184, 185, 186, 187, 188, 189, 190, 191, 192, 193, 194, 195, 196, 197, 198, 199, 200])
RX data:
array('I', [197, 198, 199, 200, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47, 48, 49, 50, 51, 52, 53, 54, 55, 56, 57, 58, 59, 60, 61, 62, 63, 64, 65, 66, 67, 68, 69, 70, 71, 72, 73, 74, 75, 76, 77, 78, 79, 80, 81, 82, 83, 84, 85, 86, 87, 88, 89, 90, 91, 92, 93, 94, 95, 96, 97, 98, 99, 100, 101, 102, 103, 104, 105, 106, 107, 108, 109, 110, 111, 112, 113, 114, 115, 116, 117, 118, 119, 120, 121, 122, 123, 124, 125, 126, 127, 128, 129, 130, 131, 132, 133, 134, 135, 136, 137, 138, 139, 140, 141, 142, 143, 144, 145, 146, 147, 148, 149, 150, 151, 152, 153, 154, 155, 156, 157, 158, 159, 160, 161, 162, 163, 164, 165, 166, 167, 168, 169, 170, 171, 172, 173, 174, 175, 176, 177, 178, 179, 180, 181, 182, 183, 184, 185, 186, 187, 188, 189, 190, 191, 192, 193, 194, 195, 196, 197, 198, 199, 200])

I checked the python module riffa.py but didn't notice anything abnormal. Could anybody advise how to fix this since my apps will base on Python.

marzoul, 2019-08-14:
@jixi2018 This is normal. The demo bitstream is built with minimal RTL code, just for timing evaluation and testing purposes. The contents of a data buffer is reused internally when switching from Rx to Tx, so the last 4 values received are sent back (for128b PCIe bitstreams).
The demo RTL is a good point to start hacking and add your own functionality, though.

gxflying, 2019-08-27:
hi, sorry to bother you, but in my riffa system, the linux just cannot find a riifa device, 
so , can you please tell me : do I need install PCIe driver for the riifa when i had installed the riifa driver ??

jixi2018, 2019-08-29:
@marzoul, Thanks for the reply. I am going to locally modify the chnl_tester.v to add a BRAM block and other logics to buffer the RX data, perform some processing. Then it will send the data in BRAM back via TX. There are RX/TX signals at the chnl_tester.v top-level. Could you let me know the timing of the following ones:

	input CHNL_RX,               //is this a pulse to start RX or should it be active during the whole RX?
	input CHNL_RX_LAST,     //same, is this a pulse asserted at the last RX word(128-bit)
	
	output CHNL_TX_LAST,          //the example set it to constant 1, it is true for all cases?