KastnerRG/riffa #51: UBUNTU18.0.4  failed to compile and install

vicmaaa, 2019-12-16:
Hi~when i use "make" and "make install" trying to compile and install riffa, a lot of errors occured:

------------------------------------------errors------------------------------------------------------------------------

/home/msl/SoftWare/riffa_2.1/source/driver/linux/riffa_driver.c:1343:2: note: (near initialization for ‘DEFINE_PCI_DEVICE_TABLE’)
In file included from /home/msl/SoftWare/riffa_2.1/source/driver/linux/riffa_driver.c:58:0:
./include/linux/pci.h:858:2: error: field name not in record or union initializer
  .vendor = (vend), .device = (dev), \
  ^
/home/msl/SoftWare/riffa_2.1/source/driver/linux/riffa_driver.c:1343:3: note: in expansion of macro ‘PCI_DEVICE’
  {PCI_DEVICE(VENDOR_ID1, PCI_ANY_ID)},
   ^~~~~~~~~~
./include/linux/pci.h:858:2: note: (near initialization for ‘DEFINE_PCI_DEVICE_TABLE’)
  .vendor = (vend), .device = (dev), \
  ^
/home/msl/SoftWare/riffa_2.1/source/driver/linux/riffa_driver.c:1343:3: note: in expansion of macro ‘PCI_DEVICE’
  {PCI_DEVICE(VENDOR_ID1, PCI_ANY_ID)},
   ^~~~~~~~~~
./include/linux/pci.h:858:12: error: invalid initializer
  .vendor = (vend), .device = (dev), \
            ^
/home/msl/SoftWare/riffa_2.1/source/driver/linux/riffa_driver.c:1343:3: note: in expansion of macro ‘PCI_DEVICE’
  {PCI_DEVICE(VENDOR_ID1, PCI_ANY_ID)},
   ^~~~~~~~~~
./include/linux/pci.h:858:12: note: (near initialization for ‘DEFINE_PCI_DEVICE_TABLE’)
  .vendor = (vend), .device = (dev), \
            ^
/home/msl/SoftWare/riffa_2.1/source/driver/linux/riffa_driver.c:1343:3: note: in expansion of macro ‘PCI_DEVICE’
  {PCI_DEVICE(VENDOR_ID1, PCI_ANY_ID)},
   ^~~~~~~~~~

------------------------------------------errors------------------------------------------------------------------------

 my os is ubuntu18.0.4, i'm really confused. 
since there is little info about this kind of problem, i have to come to you for help, if you can give me some tips, i'd be very glad~thank u

vicmaaa, 2019-12-16:
[image]

warmonkey, 2019-12-30:
https://github.com/marzoul/riffa/commit/77b91a36441b652a729c7590847e4520e5ae4c1a

closed 2024-06-11 by marzoul