KastnerRG/riffa #55: UBUNTU20.0.4 failed to execute sudo make command

qingcai52, 2021-06-24:
Hello,
 I executed "sudo make" command and report the error, I don't know what causes this error, Please help to take a look at this issue, thanks!
/////////////////////////////////////////////////////////////////////////////////////////////////////////////
host@host-System-Product-Name:/home/work/riffa_master_mz/driver/linux$ sudo make
Compiling driver for kernel: 5.8.0-59-generic with the following values
     NAME: 'riffa'
   MAJNUM: '100'
VENDOR_ID0: '10EE'
VENDOR_ID1: '1172'

sed -i 's/#define MAJOR_NUM [^\n]*/#define MAJOR_NUM 100/g' riffa_driver.h
sed -i 's/#define DEVICE_NAME [^\n]*/#define DEVICE_NAME "riffa"/g' riffa_driver.h
sed -i 's/#define VENDOR_ID0 [^\n]*/#define VENDOR_ID0 0x10EE/g' riffa_driver.h
sed -i 's/#define VENDOR_ID1 [^\n]*/#define VENDOR_ID1 0x1172/g' riffa_driver.h
sed -i 's/#define DEBUG [^\n]*/#define DBUG 1/g' riffa_driver.h
sed -i 's/#define DBUG [^\n]*/#define DBUG 1/g' riffa_driver.h
make -C /lib/modules/`uname -r`/build M=/home/work/riffa_master_mz/driver/linux modules
make[1]: 进入目录“/usr/src/linux-headers-5.8.0-59-generic”
  CC [M]  /home/work/riffa_master_mz/driver/linux/riffa_driver.o
/home/work/riffa_master_mz/driver/linux/riffa_driver.c: In function ‘fill_sg_buf’:
/home/work/riffa_master_mz/driver/linux/riffa_driver.c:449:27: error: ‘struct mm_struct’ has no member named ‘mmap_sem’; did you mean ‘mmap_base’?
  449 |   down_read(&current->mm->mmap_sem);
      |                           ^~~~~~~~
      |                           mmap_base
/home/work/riffa_master_mz/driver/linux/riffa_driver.c:457:25: error: ‘struct mm_struct’ has no member named ‘mmap_sem’; did you mean ‘mmap_base’?
  457 |   up_read(&current->mm->mmap_sem);
      |                         ^~~~~~~~
      |                         mmap_base
make[2]: *** [scripts/Makefile.build:286：/home/work/riffa_master_mz/driver/linux/riffa_driver.o] 错误 1
make[1]: *** [Makefile:1785：/home/work/riffa_master_mz/driver/linux] 错误 2
make[1]: 离开目录“/usr/src/linux-headers-5.8.0-59-generic”
make: *** [Makefile:101：riffa.ko] 错误 2

qingcai52, 2021-06-24:
[image]

qingcai52, 2021-06-24:
[image]

marzoul, 2021-06-24:
Hi @qingcai52
I updated my fork of the repo with 1 commit that fixes compilation with linux kernel versions >= 5.8, you may want to try it.
https://github.com/marzoul/riffa
Also in pull request #18

qingcai52, 2021-06-25:
Hi, marzoul, I solved the isuse follow the way you said,  thanks!

closed 2021-06-27 by qingcai52