KastnerRG/riffa #58: riffa on ubuntu

erginatalar, 2023-04-17:
I am new to driver installations. Can someone help me with the installation issue?  I tried to install riffa on my ubuntu system and received the following error message:

erginatalar@AtalarUbuntu:~/Programming/riffa-master/driver/linux$ sudo make debug
Compiling driver for kernel: 5.19.0-38-generic with the following values
     NAME: 'riffa'
   MAJNUM: '100'
VENDOR_ID0: '10EE'
VENDOR_ID1: '1172'

sed -i 's/#define MAJOR_NUM [^\n]*/#define MAJOR_NUM 100/g' riffa_driver.h
sed -i 's/#define DEVICE_NAME [^\n]*/#define DEVICE_NAME "riffa"/g' riffa_driver.h
sed -i 's/#define VENDOR_ID0 [^\n]*/#define VENDOR_ID0 0x10EE/g' riffa_driver.h
sed -i 's/#define VENDOR_ID1 [^\n]*/#define VENDOR_ID1 0x1172/g' riffa_driver.h
sed -i 's/#define DEBUG [^\n]*/#define DBUG 1/g' riffa_driver.h
sed -i 's/#define DBUG [^\n]*/#define DEBUG 1/g' riffa_driver.h
make -C /lib/modules/`uname -r`/build SUBDIRS=`pwd` modules
make[1]: Entering directory '/usr/src/linux-headers-5.19.0-38-generic'
make[2]: *** No rule to make target 'arch/x86/entry/syscalls/syscall_32.tbl', needed by 'arch/x86/include/generated/uapi/asm/unistd_32.h'.  Stop.
make[1]: *** [arch/x86/Makefile:232: archheaders] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.19.0-38-generic'
make: *** [Makefile:101: riffa.ko] Error 2

marzoul, 2023-04-18:
Hi,
I have never tried the `make debug` target, this compilation recipe may lack maintenance.
It is probably not what you want to just install the driver, just use `make`, no `sudo` should be needed for compilation.
Or, maybe you have not installed the Linux kernel headers package ?

Arnoud-Bleeker, 2023-07-26:
I had a similar issue changing `make -C $(KDIR) SUBDIRS=pwd modules` to `make -C $(KDIR) M=$(shell pwd) modules` in the makefile solved it.

marzoul, 2023-09-21:
I've just realized that the proposed fix is nearly exactly the patch I did 2 years ago for the Archlinux user package :sweat_smile: 
At that time I had no visibility about whether or not this change would be appropriate for other distros too.
https://aur.archlinux.org/packages/riffa-git
@Arnoud-Bleeker if you want to author a PR you would be welcome. Or I can do it in a few days.

bartokon, 2024-01-05:
Ubuntu 22.04.1
`Compiling driver for kernel: 6.2.0-39-generic with the following values
     NAME: 'riffa'
   MAJNUM: '100'
VENDOR_ID0: '10EE'
VENDOR_ID1: '1172'

sed -i 's/#define MAJOR_NUM [^\n]*/#define MAJOR_NUM 100/g' riffa_driver.h
sed -i 's/#define DEVICE_NAME [^\n]*/#define DEVICE_NAME "riffa"/g' riffa_driver.h
sed -i 's/#define VENDOR_ID0 [^\n]*/#define VENDOR_ID0 0x10EE/g' riffa_driver.h
sed -i 's/#define VENDOR_ID1 [^\n]*/#define VENDOR_ID1 0x1172/g' riffa_driver.h
sed -i 's/#define DEBUG [^\n]*/#define DBUG 1/g' riffa_driver.h
sed -i 's/#define DBUG [^\n]*/#define DBUG 1/g' riffa_driver.h
echo "make -C /lib/modules/`uname -r`/build SUBDIRS=`pwd` modules"
make -C /lib/modules/6.2.0-39-generic/build SUBDIRS=/home/bartek/riffa/driver/linux modules
make -C /lib/modules/`uname -r`/build M=/home/bartek/riffa/driver/linux modules
make[1]: Entering directory '/usr/src/linux-headers-6.2.0-39-generic'
  CC [M]  /home/bartek/riffa/driver/linux/riffa_driver.o
/home/bartek/riffa/driver/linux/riffa_driver.c: In function ‘fill_sg_buf’:
/home/bartek/riffa/driver/linux/riffa_driver.c:469:29: error: too few arguments to function ‘get_user_pages’
  469 |                 num_pages = get_user_pages(udata, num_pages_reqd, FOLL_WRITE, pages);
      |                             ^~~~~~~~~~~~~~
In file included from ./include/linux/scatterlist.h:8,
                 from ./include/linux/dmapool.h:14,
                 from ./include/linux/pci.h:1555,
                 from /home/bartek/riffa/driver/linux/riffa_driver.c:52:
./include/linux/mm.h:2129:6: note: declared here
 2129 | long get_user_pages(unsigned long start, unsigned long nr_pages,
      |      ^~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:260: /home/bartek/riffa/driver/linux/riffa_driver.o] Error 1
make[1]: *** [Makefile:2026: /home/bartek/riffa/driver/linux] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.2.0-39-generic'
make: *** [Makefile:102: riffa.ko] Error 2
`
makefile change:
`	echo "make -C $(KDIR) SUBDIRS=`pwd` modules"
	make -C $(KDIR) M=$(shell pwd) modules
` what can be the cause?

marzoul, 2024-01-05:
Hi, I prepared fixes some time ago to compile with linux 6.5.
Your issue with `get_user_pages` seems part of what I fixed on my side.
I am not sure exactly what versions of linux introduced the changes.
https://github.com/KastnerRG/riffa/compare/master...marzoul:riffa:linux-65

bartokon, 2024-01-05:
Yea but I'm already using code from "master" branch so it already includes your fix. Idk why but it seems like #define if else doesn't detect kernel version correctly. Linux version code.... how to get that number?
bartek@bartek-MS-7D25:~/riffa/driver/linux$ uname -a
Linux bartek-MS-7D25 6.2.0-39-generic #40~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Nov 16 10:53:04 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
and shall I change it in version.h?

marzoul, 2024-01-05:
You are compiling for linux 6.2 so it is worth trying to lower the version "threshold" in the code of `riffa_driver.c` near line 466 probably. Change this :
`#elsif LINUX_VERSION_CODE < KERNEL_VERSION(6,5,0)`
into this :
`#elsif LINUX_VERSION_CODE < KERNEL_VERSION(6,2,0)`

marzoul, 2024-06-04:
Indeed. Quite surprising that there is no error nor warning on my side.
I will fix this (and the shell thing in the Makefile which got forgotten).

marzoul, 2024-06-05:
Pull request opened in #64 
Please comment about any missing fix.

lixiaohuan008, 2024-07-07:
Linux lubancat 5.10.160 

root@lubancat:/home/cat/riffa_test/source/driver/linux# make
Compiling driver for kernel: 5.10.160 with the following values
     NAME: 'riffa'
   MAJNUM: '100'
VENDOR_ID0: '10EE'
VENDOR_ID1: '1172'

sed -i 's/#define MAJOR_NUM [^\n]*/#define MAJOR_NUM 100/g' riffa_driver.h
sed -i 's/#define DEVICE_NAME [^\n]*/#define DEVICE_NAME "riffa"/g' riffa_driver.h
sed -i 's/#define VENDOR_ID0 [^\n]*/#define VENDOR_ID0 0x10EE/g' riffa_driver.h
sed -i 's/#define VENDOR_ID1 [^\n]*/#define VENDOR_ID1 0x1172/g' riffa_driver.h
sed -i 's/#define DEBUG [^\n]*/#define DBUG 1/g' riffa_driver.h
sed -i 's/#define DBUG [^\n]*/#define DBUG 1/g' riffa_driver.h
make -C /lib/modules/`uname -r`/build SUBDIRS='pwd' modules
make[1]: Entering directory “/usr/src/linux-headers-5.10.160”
/usr/bin/env: “python”: No such file or directory
  SYNC    include/config/auto.conf.cmd
scripts/kconfig/conf: 1: Syntax error: "(" unexpected
make[3]: *** [scripts/kconfig/Makefile:71：syncconfig] Error 2
make[2]: *** [Makefile:630：syncconfig] Error 2
make[1]: *** [Makefile:750：include/config/auto.conf.cmd] Error 2
make[1]: Leaving directory “/usr/src/linux-headers-5.10.160”
make: *** [Makefile:101：riffa.ko] Error 2

I complied Riffa2.2.2 under my ARM developed kit board which is flashed Linux lubancat 5.10.160 image. The complied environment is a image not a linux kernel source SDK, please help to locate the cause of the errors above. I am not sure if the compilation error is due to the environment.  Thanks.

marzoul, 2024-08-20:
I suppose the issue is fixed so I am closing it.
Please reopen if necessary.

closed 2024-08-20 by marzoul